fluidd-core/fluidd #1459: Unable to close/clear notifications panel

thijstriemstra, 2024-06-30:
### Fluidd Version

1.30.1

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

I'm unable to use "Clear all" (greyed out) this message and the "x" close button is also not working, clicking elsewhere on the screen hides the panel.

[image]

### What did you expect to happen

Being able to clear the message and close the panel.

### How to reproduce

See image above.

thijstriemstra, 2024-10-21:
can someone acknowledge this bug report ?

pedrolamas, 2024-10-22:
Hi @thijstriemstra, thank you for reporting this.

I've taken a look at it and this behavior is actually "by design"!

We keep the "previously throttled" warning hidden on the notification pop-up, but it is only shown when you actually open it (there's no other indication elsewhere)

Having said that, one can argue that it would make sense to allow to clear it as we do with other notifications... I personally don't have any strong opinion on either way.

thijstriemstra, 2024-10-23:
Thanks for the feedback. Could you also reproduce "the "x" close button is also not working"?

pedrolamas, 2024-10-23:
> Thanks for the feedback. Could you reproduce "the "x" close button is also not working"?

Yes, as we don't allow clearing the "previously frequency capped" message (no "close" button on it) then we also disable the "clear all" option, once again, by design.

pedrolamas, 2024-11-10:
Closing this as the behavior is "by design"

closed 2024-11-10 by pedrolamas