fluidd-core/fluidd #1460: Temperature sensor name incorrect

thijstriemstra, 2024-06-30:
### Fluidd Version

1.30.1

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

I defined:

```
[temperature_sensor RPI_3B+]
sensor_type: temperature_host
min_temp: 0
max_temp: 100
```

And it shows up as `RPI 3 B`:

[image]

### What did you expect to happen

That it shows up as 'RPI 3B+' (like KlipperScreen does).

### How to reproduce

Define:

```
[temperature_sensor RPI_3B+]
sensor_type: temperature_host
min_temp: 0
max_temp: 100
```

MPFGlaser, 2024-07-17:
Good catch. Facing the same issue, but didn’t find yours since I searched for “spaces” and made my own. Good to see they’re merged now, though!

thijstriemstra, 2024-07-17:
Maybe you can copy paste your original report here as well, had a lot of extra info @MPFGlaser

PR #1468 (2024-07-26): fix: item name format and capitalization
Fluidd item names/labels format currently removes any non-letter/digit and adds spaces between then, which is incorrect.

The new approach works like this:

- Replace any `_` with ` ` (single space)
- Split string by single space
- Remove any empty entry (resulted from multiple consecutive spaces)
- Capitalize the first letter of each part of the splited string
- Join the result, separating the items with a single space

## Before

[image]

## After

[image]

Fixes #1460
closed 2024-07-26 by pedrolamas