fluidd-core/fluidd #1463: Incorrect URLs in Software Update section

rmcfadzean, 2024-07-08:
### Fluidd Version

all

### Browser

Chrome, Firefox, Safari, Edge

### Device

Desktop PC, Laptop, Mobile device (Phone, Tablet or similar)

### Operating System

Windows, macOS, Linux, Android, iOS

### What happened

In the `Software Updates` section of the `Settings` UI,
When the user has opened the Commit History modal for a versioned package type (`git_repo`, `zip`, `web`),
The URL for both the author's name, and the commit hash are not based on the repo's URL.

Using the config below, the URLs produced are  `//github.com/owner/package-key[/commit/$sha]`

```
[update_manager package-key]
type: git_repo|web|zip
origin: https://github.com/owner/klipper-plugin_name.git
```

[image]
[image]

### What did you expect to happen

The URL(s) should have been `//github.com/owner/klipper-plugin_name[/commit/$sha]`

### How to reproduce

- Add a `git_repo` type `[update_manager pkg]` section to your Moonraker config
- Ensure it is a commit/version lower than the current so that Moonraker picks up there's an update
- Restart Moonraker
- Click `Check for Updates` in the `Software Updates` section of the `Settings` UI
- Click the icon on the right of that package's row to pop the `Commit History` modal
- Inspect any of the links inside of the modal

### Additional information

I understand that most related packages/apps are installed via helper scripts like Kiauh or an OS like RatOS, and in general, they do seem to match the `key` with the full `repo_name` so this could be considered a config issue rather than a bug.

With that said though, Moonraker's API does return the `repo_name` along with the `owner` which is enough rebuild (all?) web UIs for git repos.

One side note too: The link to the right of the package name links directly to the `remote_url` which is fine for GitHub HTTPS links but this could also be changed to the `owner/repo_name` structure.

This is an example of the Moonraker API response for `moonraker:7125/machine/update/status` including the `repo_name` key.

```json
{
  "led-effect": {
    "channel": "dev",
    "debug_enabled": false,
    "channel_invalid": false,
    "is_valid": true,
    "configured_type": "git_repo",
    "info_tags": [],
    "detected_type": "git_repo",
    "remote_alias": "origin",
    "branch": "master",
    "owner": "julianschill",
    "repo_name": "klipper-led_effect",
    "remote_url": "https://github.com/julianschill/klipper-led_effect.git",
    "recovery_url": "https://github.com/julianschill/klipper-led_effect.git",
    "version": "v0.0.11-40",
    "remote_version": "v0.0.12-0",
    "rollback_version": "v0.0.11-40",
    "current_hash": "28a3c25220deba27b35b15e96d4dae3b761cdec1",
    "remote_hash": "03a46eab5670c2934b38093993834f822a56c0fe",
    "is_dirty": false,
    "detached": false,
    "commits_behind": [
      {
        "sha": "03a46eab5670c2934b38093993834f822a56c0fe",
        "author": "Julian Schill",
        "date": "1719865580",
        "subject": "rename setup_minmax to setup_adc_sample for analog effect. Fixes [BUG]  'MCU_adc' object has no attribute 'setup_minmax' #211",
        "message": "",
        "tag": "v0.0.12"
      }
    ],
    "commits_behind_count": 1,
    "git_messages": [],
    "full_version_string": "v0.0.11-40-g28a3c252",
    "pristine": true,
    "corrupt": false,
    "warnings": [],
    "anomalies": []
  }
}
```

pedrolamas, 2024-07-08:
Hi @rmcfadzean, thank you for opening this ticket.

I am not sure I fully understand the problem here, can you provide a concrete snippet of `[update_manager led-effect]` section you are using and how is the folder structure for this repo in your file system?

pedrolamas, 2024-07-08:
> I understand that most related packages/apps are installed via helper scripts like Kiauh or an OS like RatOS, and in general, they do seem to match the `key` with the full `repo_name` so this could be considered a config issue rather than a bug.

Ok, this cleared things now, and yes, you are correct! We should use `repo_name` there, as you are doing in your PR.

rmcfadzean, 2024-07-08:
Awesome! :tada: 

Apologies about not including an actual config. For reference, here's the one I thought I included:

```yaml
[update_manager led-effect]
type: git_repo
path: ~/led-effect
origin: https://github.com/julianschill/klipper-led_effect.git
is_system_service: False
```

My local version is on commit https://github.com/julianschill/klipper-led_effect/commit/28a3c25220deba27b35b15e96d4dae3b761cdec1, 1 commit behind master @ https://github.com/julianschill/klipper-led_effect/commit/03a46eab5670c2934b38093993834f822a56c0fe

PR #1464 (2024-07-28): feat: Fix repo URLs in version information dialog
Use `owner/repo_name` instead of `owner/key` for the repo URLs in the version information dialog.

Updated related types.

Fixes #1463

As mentioned in #1463, I could also update the URL in that the "External Link" on the `Software Updates` settings UI.
closed 2024-07-28 by matmen