fluidd-core/fluidd #1466: Fluidd adds spaces between characters of the names in the "Thermals" panel

MPFGlaser, 2024-07-16:
### Fluidd Version

v1.30.1

### Browser

Firefox

### Device

Laptop

### Operating System

macOS

### What happened

In the Thermals panel, spaces get added to the names of the various thermistors I have on my printer. See screenshots below of the interface and the `printer.cfg`  
The spaces disappear if you use a string that is of the same "kind" (i.e., lower case, upper case, numbers. See below.)

[image]
[image]

### What did you expect to happen

I expected no spaces to show up, since I did not add them into the name in the `printer.cfg`.

### How to reproduce

1. Have a temperature sensor
2. Use Fluidd
3. Name it something where the "type" of character changes, e.g. `M8P`
4. You will see `M 8 P`

### Additional information

This only happens when you mix the character types.
Examples:
| `printer.cfg` | Fluidd |
| --- | --- |
| `SoC` | `S o C` |
| `SOC` | `SOC`|
| `CB1` | `CB 1` |
| `M8P` | `M 8 P` |

It's a very minor thing, but it's bugging me nonetheless. ^^
Thanks in advance!

thijstriemstra, 2024-07-17:
i reported same in #1460 already.

pedrolamas, 2024-07-17:
Duplicate of #1460