fluidd-core/fluidd #1473: Web interface not visible

T9Air, 2024-08-09:
### Fluidd Version

v1.30.1

### Browser

Chrome

### Device

Laptop

### Operating System

Windows

### What happened

If I try to go to fluidd in my browser, it looks like it is fully loaded, with the logo at the top of the tab, etc. but the browser page is blank, with no UI visible. I even tried reinstalling Fluidd, and it did not help

### What did you expect to happen

I expect to be able to see and be able to interact with the Fluidd web interface

### How to reproduce

I think I changed the trusted_clients option in the moonraker.conf to my laptops IP address, but it did work after that. Then it randomly stopped working. When I tried to reinstall it, it still did not work

franco443, 2024-08-11:
try to clear your browser cache. That did the trick for me on browser. I am still trying to figure orcaslicer out its doing that to me there for fluidd interface. That should work for you though on browser. I have chrome too. Hope that works for you.

T9Air, 2024-08-11:
I tried that, didn't work.

T9Air, 2024-08-12:
@pedrolamas I reset all of my changes, and it was still showing nothing.

pedrolamas, 2024-08-12:
In that case, you might want to press F12 to open the in-browser dev tools and check the console there for errors

T9Air, 2024-08-12:
I get this error: `index-DmW-YI4h.js:254 WebSocket connection to 'ws://3dprintermanager:7125/websocket' failed: WebSocket is closed before the connection is established.` Do you know how I could fix it?

T9Air, 2024-08-12:
Mainsail does work if that would change anything.

pedrolamas, 2024-08-12:
Please upload your moonraker.log here, that might help to understand what is going on.

T9Air, 2024-08-12:
[moonraker.log](https://github.com/user-attachments/files/16588970/moonraker.log)

grilparto, 2024-08-14:
Wanted to add another datapoint.  I'm having the same issue.  It occurred after trying to adjust my layout.  I experienced some odd behavior upon exiting the layout adjustment mode.  Each column on the UI was showing a mix of the actual UI element and also the layout adjustment behavior.  After refreshing the page, blank (white) page as described above.  Also seeing the same "failed: WebSocket is closed before the connection is established." in my Chrome console (given that we're both having the same blank page issue and websocket failure, it seemed like both our issues could be related).

Tried clearing website data, tried a different browser, tried a browser on a different PC also.  Removed and reinstalled fluidd, then removed fluidd, removed moonraker, and then reinstalled both of those as well.  Also tried removing/reinstalling fluidd and moonraker again, but also deleting the moonraker.conf to make sure a new one would be created after the reinstall.  

Also attaching my moonraker.log

[moonraker.log](https://github.com/user-attachments/files/16619076/moonraker.log)

pedrolamas, 2024-08-15:
Can I ask that you access http://moonraker-ip-or-host:7125/server/database/item?namespace=fluidd (replacing `moonraker-ip-or-host` with the correct IP or hostname for the moonraker instance) and paste the results here?

We've previously had reports of similar issues with the dashboard causing the page to go blank, and this might help to get to the bottom of this issue.

T9Air, 2024-08-15:
[fluidd.txt](https://github.com/user-attachments/files/16625587/fluidd.txt)

pedrolamas, 2024-08-15:
Thank you @T9Air, I can see the problem clearly on that and I will provide a fix for next release!

PR #1484 (2024-08-16): fix: null check cards on layout init
We have had a few reports from users that Fluidd stops working after editing the dashboard layout.

It seems that somehow the layout data is getting incorrectly recorded in the database with `null` values for some cards.

[image]

Though I don't yet know the reason for this happening, the proposed fix here is to filter these `null` values out before we attempt to load the layout.

Fixes #1473
closed 2024-08-16 by pedrolamas