fluidd-core/fluidd #1475: Fluidd keyboard shortcuts take precedence over OS shortcuts

bwnance, 2024-08-09:
### Fluidd Version

v1.30.1

### Browser

Chrome

### Device

Laptop

### Operating System

macOS

### What happened

OS-level shortcuts are overridden by the new keyboard shortcuts system.
Primarily, this is annoying when trying to copy text via cmd + c.

### What did you expect to happen

cmd + c should copy text, not take me to the console

### How to reproduce

select text (while not in the console tab) and press cmd + c - the text won't be copied, and you'll be moved to the console tab.

pedrolamas, 2024-08-12:
Hi @bwnance, thank you for opening this issue.

To be honest this was mostly due to my ignorance to how mac keyboards work than anything else... we will try to get this fixed for the next release.

bwnance, 2024-08-12:
> Hi @bwnance, thank you for opening this issue.
> 
> To be honest this was mostly due to my ignorance to how mac keyboards work than anything else... we will try to get this fixed for the next release.

Thank you! 🙏

PR #1477 (2024-08-12): fix: Check for Meta key on keyboard shortcuts
Check for Meta key press (<kbd>Cmd</kbd> key in Mac, <kbd>Win</kbd> key in Windows)

Fixes #1475
closed 2024-08-12 by pedrolamas