fluidd-core/fluidd #1490: G-Code Previw Show Very Drastically incorrect number of layers - File Exported from Cura Ultimaker 5.8

soundtechscott, 2024-08-29:
### Fluidd Version

v0.12.0-102-g9f41f53c

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

I builld a file in Cura Ultimaker 5.8
slice, export to G-code or Upload directly to 3d printer
Fluidd shows correct number of layers in Printing window

but DRASTICALLY incorrect number of layers in the G-Code Preview window

[image]

[EN4PRO_ Front Link.zip](https://github.com/user-attachments/files/16808366/EN4PRO_.Front.Link.zip)
[EN4PRO_Front Link.zip](https://github.com/user-attachments/files/16808367/EN4PRO_Front.Link.zip)

### What did you expect to happen

I expect it to match....

### How to reproduce

slice, upload, look 

### Additional information

this worked just fine in a previous version of fluidd
and I dont know why it doesnt work now

pedrolamas, 2024-08-30:
Hi @soundtechscott, thank you for opening this ticket.

This is actually expected behavior as Cura does not directly provide layer information to Klipper; my recommendation is that you use my [Klipper Preprocessor script for Cura](https://github.com/pedrolamas/klipper-preprocessor) to ensure that all sliced files have the necessary data.

closed 2024-08-30 by pedrolamas

soundtechscott, 2024-08-30:
this worked fine in cura 5.7
what has changed with fluidd/klipper that has now made cura not work

pedrolamas, 2024-08-30:
> what has changed with fluidd/klipper that has now made cura not work

Nothing changed in Fluidd or Klipper side, it might have changed in Cura 5.8, but that I cannot say... nevertheless, both Fluidd and Mainsail have always recommended using that script in Cura to ensure that the output g-code is fully compatible with the Klipper ecosystem.

pedrolamas, 2024-08-30:
@soundtechscott I just realized that on the initial post you indicated you are using Fluidd v0.12.0, which is more than 3 years old!

The odd bit is that I can't find the full version "v0.12.0-102-g9f41f53c", so maybe you are actually using a fork (altered copy) of Fluidd?

soundtechscott, 2024-08-30:
installed straight from Kiauh
I'm only just now 1 version off according to this

[image]

soundtechscott, 2024-08-30:
and inside Fluidd
[image]

soundtechscott, 2024-08-30:
also
I am up to date according to these links

https://www.klipper3d.org/Releases.html
https://github.com/fluidd-core/fluidd/releases

I switched to the octopus pro board on my neptune to get away from a "Fork" and have the standard klipper
because elegoo does stupid things in their hacked up firmware

pedrolamas, 2024-08-30:
> and inside Fluidd [...]

Ah, you indicated the Klipper version, not Fluidd which is what was asked!

All explained then, and yes, you are using the most recent version of Fluidd! 🙂

soundtechscott, 2024-08-30:
also, Cura claiming its your problem.....
so.. ????

[image]

pedrolamas, 2024-08-30:
They are correct about one thing: Cura does not produce the best g-code for Klipper, hence why I created that script and it has been promoted a helper tool.

Bottom line is that you should use the script above if you plan to slice files in Cura and then print them in a Klipper powered printer.