fluidd-core/fluidd #1491: go2rtc api uri broken

aam335, 2024-08-31:
### Fluidd Version

v1.30.3

### Browser

Chrome

### Device

Desktop PC, Laptop

### Operating System

Linux

### What happened

ws url for go2rtc webrtc broken:
ws://rockpi.local:1984/api/api/ws?src=cam

### What did you expect to happen

ws://rockpi.local:1984/api/ws?src=cam

### How to reproduce

select go2rtc camera source 

### Additional information

possibly you need to add origin any to go2rtc config
```
api:
  origin: "*"
```

pedrolamas, 2024-08-31:
Hi @aam335, thank you for opening this ticket.

Can you confirm that you did indeed add this section to your go2rtc configuration?

```
api:
  origin: "*"
```

I ask this as you DO need to whitelist the domain, without that the browser will block the access with a CORS error!

aam335, 2024-08-31:
> I ask this as you DO need to whitelist the domain, without that the browser will block the access with a CORS error!
@pedrolamas  it is not a CORS problem, go2rtc limits access to signalling ws to known referers.
https://github.com/AlexxIT/go2rtc/issues/118

pedrolamas, 2024-08-31:
Sorry, but I just tested and can confirm that it does work if you do add the "api/origin" configuration.

[image]

If I remove the final 2 lines, the stream does not load, but once I add it, it works fine!

pedrolamas, 2024-08-31:
What URL are you using? I just copied this one and used it in Fluidd!

[image]

aam335, 2024-08-31:
http://rockpi.local:1984/stream.html?src=cam&mode=webrtc
http://rockpi.local:1984/stream.html?src=cam

pedrolamas, 2024-08-31:
I used the 2nd URL you have there and added a test camera on my system:

[image]

That seems to load the correct URL: 

[image]

aam335, 2024-08-31:
I confused snapshot and stream url )

http://rockpi.local:1984/api/frame.jpeg?src=cam

adds extra api

thanks!!!

closed 2024-08-31 by aam335