fluidd-core/fluidd #1497: /#/login page creates confusion when user is trusted / authorized

crabdancing, 2024-09-24:
### Fluidd Version

v1.30.0-f3e4ac3

### Browser

Chrome

### Device

Desktop PC, Laptop

### Operating System

Linux

### What happened

When using moonraker+fluidd and accessing over 127.0.0.1 (trusted user), if the user somehow ends up on `http://127:0.0.1/#/login`, they are presented with a login screen, which can easily confuse the user into thinking they are locked out of the account. If you go to `http://127.0.0.1/#` however, the login screen disappears.

### What did you expect to happen

Expect redirect from `/#/login` to /#/` before reaching login page, if the user is already authorized.

### How to reproduce

No accounts. Make sure 127.0.0.1 is configured as trusted. Can use SSH reverse tunnel if running klipper/fluidd/moonraker stack remotely. Connect to `http://127.0.0.1/#/login`. Notice that login page appears needlessly. The correct behavior would be to redirect to `http://127.0.0.1/#` because the user is already logged in.

### Additional information

Issue moved over from [moonraker's tracking](https://github.com/Arksine/moonraker/issues/907).

closed 2024-09-24 by pedrolamas

crabdancing, 2024-09-24:
Thanks for the quick fix, @pedrolamas! :D

pedrolamas, 2024-09-24:
Thank you for opening the issue @crabdancing!

I had noticed that problem a few times myself, but sometimes this is the push we need to put the work in and get a fix out! 🙂

PR #1498 (2024-09-24): fix: redirect from login page if authenticated
Moonraker provides an authentication feature but has no authorization concept, so when users are authorized as an trusted (anonymous) user, there is no point showing the login page, and thus safe to redirect to main page, which is what this PR fixes.

Fixes #1497