fluidd-core/fluidd #1501: Console output messed up

Rigattoni, 2024-10-02:
### Fluidd Version

v1.31.0-37 

### Browser

Firefox

### Device

Laptop

### Operating System

Windows

### What happened

I found a display issue when printing help command in the console. It is shown fine in the console tab.
Further more: The delete button in the small console does not work correctly. At least if you have the screen as shown in the pic
[image]

### What did you expect to happen

No overlapping of different text parts

### How to reproduce

Type help in the console on the main screen.

HelgeKeck, 2024-10-02:
this is the wrong github repo, please repost it here
https://github.com/fluidd-core/fluidd

HelgeKeck, 2024-10-02:
i had this as well once or twice, it happens always when we output some html code, not sure if this is related to the mainline, thats why i suggest to repost it and close this one here

pedrolamas, 2024-10-02:
Closing now as this is not a mainline Fluidd issue.

closed 2024-10-02 by pedrolamas