fluidd-core/fluidd #1503: jobs: hide .mru files

thijstriemstra, 2024-10-08:
### Fluidd Version

1.30.4

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

The jobs page has `.mru` files in it:

[image]

I enabled the hidden files filter.

### What did you expect to happen

That .mru files are hidden.

### How to reproduce

Not sure..!

thijstriemstra, 2024-10-10:
my guess is mru stands for moonraker upload? i'll try to search moonraker codebase.

thijstriemstra, 2024-10-10:
my guess is they're unprocessed or failed moonraker uploads? not sure how they ended up in this state though.

closed 2024-10-11 by pedrolamas

thijstriemstra, 2024-10-21:
.> my guess is they're unprocessed or failed moonraker uploads? not sure how they ended up in this state though.

cc @arksine

Arksine, 2024-10-22:
Moonraker streams uploads to the system temp folder with a temporary file name (with the .mru extension).  Once the upload is complete, the file is moved to the destination path and renamed.  The only way I'm aware that a `.mru` file could be moved  without being renamed is if the upload request either omitted the multipart filename or passed an empty value.  Its very unlikely that Fluidd did this unless its a browser bug, however it may be possible that some other software did.

I added a check to the upload method to make sure it receives a valid file name.  That said, if this behavior continues `moonraker.log` might be useful in tracking down what would cause this.

thijstriemstra, 2024-10-23:
thanks for the feedback @Arksine!

thijstriemstra, 2024-11-19:
> That said, if this behavior continues moonraker.log might be useful in tracking down what would cause this.

@Arksine I got the new 400 error you added recently while uploading a file from OrcaSlicer. It showed up in a OrcaSlicer popup:

```
Error uploading to print host:
dd0b:319:6e26::130: curl:Couldn't connect to server:
Failed to connect to dd0b:319:6e26::130 port 80: Connection refused
[Error 7]
192.168.2.181: HTTP 400: {"error":{"code":400,"message":"Bad Request","traceback":"Traceback
(most recent call last):\n\n File
\"/home/foo/moonraker-env/lib/python3.11/site-packages/tornado/web.py\", line 1790, in
_execute\n result = await result\n ^^^^^^^^^^^^\n\n File
\"/home/foo/moonraker/moonraker/components/application.py\", line 1001, in post\n raise
tornado.web.HTTPError(400, \"Multipart filename omitted\")\n\ntornado.web.HTTPError: HTTP
400: Bad Request (Multipart filename omitted)\n"}}
```

I uploaded a file with title: `Z-Stepper Spacer_ASA_23m6s.gcode` It would fail everytime I uploaded.

After changing the name to `Z-Stepper_Spacer_ASA_23m6s.gcode`, thus eliminating the space in filename, it worked properly.

So where is the issue, Moonraker or OrcaSlicer?

Arksine, 2024-11-19:
Sounds like an issue in Orcaslicer.  Its uploading a file without properly setting the filename in the `Content-Disposition` header.

FWIW, I uploaded a file with PrusaSlicer 2.8.1 using the same file name and it proceeded successfully.

PR #1507 (2024-10-11): feat(FileSystem): allow filtering Moonraker temp upload files
Adds a new optional file system filter for Moonraker temporary upload files (.mru)

[image]

Resolves #1503