fluidd-core/fluidd #1510: Configuration changes do not save

cumminsrover, 2024-10-16:
### Fluidd Version

v1.30.4

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

I upgraded to the latest Fluidd, Moonraker, Klipper, etc.

When I edit the printer.cfg through Fluidd, and choose "SAVE & RESTART" or "SAVE", 19 times out of 20, Fluidd goes through the process but does NOT actually save the changes.

Fluidd indicates that the change has worked, yet the printer still fails to start because the config file changes did not actually happen. Refreshing the configuration files shows that the files were not actually updated.

### What did you expect to happen

Choosing "SAVE" or "SAVE & CLOSE" from the editor should actually save the files.

### How to reproduce

Confirmed on Windows 11, 10, Ubuntu 24.04, 22.04, RaspberryPi OS 10, 11, 12

1. Open Fluidd interface
2. Navigate to Configuration
3. Edit a configuration file
4. Choose "SAVE & CLOSE"
5. Verify that timestamp is not actually updated in edited file

### Additional information

I have not tried other browsers yet.

I am not sure how to get any actual useful debugging on this problem as there are no errors generated in the fluidd-error.log

Thank you!

pedrolamas, 2024-10-16:
Hi @cumminsrover, thank you for opening this issue.

Fluidd is just a web interface that runs in your browser; Moonraker is the component that actually does the work of loading and saving files.

Can you please upload your moonraker.log so I can check it for issues?

cumminsrover, 2024-10-16:
Oh, sorry, I was a bit confused how everything worked and thought it was Fluidd...

If I need to go post it over at Moonraker, I can, just let me know please.

[moonraker.2024-10-15.log](https://github.com/user-attachments/files/17401551/moonraker.2024-10-15.log)
[moonraker.log](https://github.com/user-attachments/files/17401545/moonraker.log)

pedrolamas, 2024-10-16:
Considering how you have the Moonraker update manager sections configured, this looks like a really old install...

Can I ask how did you setup your system? Did you use Kiauh to install Klipper, Moonraker and Fluidd?

cumminsrover, 2024-10-16:
It was installed following the Voron 2.4 tutorials in the fall of 2021. It was basically a manual install at that point, KIAUH was around at the time, but I wasn't aware of it.

I'm not super excited about doing a fresh install. I only upgraded to get cancel object :/

pedrolamas, 2024-10-16:
Ah, so this is a FluiddPi install I guess... We have deprecated it a long time ago and are aware of quite a few issues with this old Raspbian.

I'm not sure why you are having this issue, but I believe it to be more related to the old OS than it being a bug in either Fluidd or Moonraker.

My advice is that you backup your printer configuration (and Moonraker database, if you want to keep the layout settings and print history) and start over.

If you have a 2nd SD-Card around you can just use that to start over and keep the existing one as a backup just in case you need to go back for any reason!

cumminsrover, 2024-10-16:
Ok, thank you Sir!

I do have extra cards around and will have to see what I can do....

pedrolamas, 2024-10-16:
Feel free to jump on [Klipper Discord](https://discord.klipper3d.org) is you run into issues and need some help - there are a lot of users there always willing to give a hand! 🙂

closed 2024-10-16 by pedrolamas