fluidd-core/fluidd #1518: No confirmation/tooltip for history delete

thijstriemstra, 2024-11-02:
### Fluidd Version

1.30.5

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

- Removing an item from history doesn't require a confirmation, i think it should (i accidentally removed something just now that i didnt want to)
- A tooltip for the remove button would be nice as well ("remove from history")

### What did you expect to happen

require confirmation to delete (or make it a default setting).

### How to reproduce

See above

thijstriemstra, 2024-11-10:
> A tooltip for the remove button would be nice as well ("remove from history")

i dont see this in 1.30.6 but might be a deliberate choice?

pedrolamas, 2024-11-11:
Granted, I didn't add a tooltip but the confirmation dialog should be showing on v1.30.6!