fluidd-core/fluidd #1545: Navigate to dashboard when adjusting layout

thijstriemstra, 2024-12-04:
### Fluidd Version

1.31.1

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

When you "adjust dashboard layout" on a page other than the dashboard, it'll show 2 buttons but there's nothing to adjust:

[image]

### What did you expect to happen

To navigate to the dashboard when clicking "adjust dashboard layout"

### How to reproduce

yep

pedrolamas, 2024-12-04:
Hi @thijstriemstra thanks for opening this issue.

Currently there are 2 places where one can edit the layout: the Dashboard and the Diagnostics (when enabled)

I guess we can disable that button on the other pages, so we'll make sure that happens for the next release.

thijstriemstra, 2024-12-04:
I guess that works, but ideally the whole layout edit thing is moved out of the menu and placed somewhere on each section where it's relevant. It took me a while before I discovered the layout edit feature, it's sort of hidden in the menu right now.

PR #1546 (2024-12-05): feat: adds Adjust Dashboard Layout sidebar button
Adds a new Adjust Dashboard Layout button to the navigation sidebar, only visible on desktop mode (will not show on mobile)

The button will also only show for the Home and the Diagnostics pages as these are the only 2 editable dashboards we currently have (I've also applied this change to the existing Adjust Dashboard Layout on the right sidebar)

## Preview

[image]

Resolves #1545
closed 2024-12-05 by pedrolamas