fluidd-core/fluidd #1553: Camera setup not working

Akshar062, 2024-12-19:
### Fluidd Version

v1.31.2

### Browser

Chrome

### Device

Desktop PC, Laptop, Mobile device (Phone, Tablet or similar)

### Operating System

Windows

### What happened

When i added camera and set camera i got invalid camera rotation snackbar and then camera preview came  but after refresh camera was gone

### What did you expect to happen

It should save camera state

### How to reproduce

just open chrome type ip and go to setting and set camera just give it name and then save it will show error then go to home there will be camera preview and refresh page camera is gone from dashboard and home also

### Additional information

I am using Elegoo neptune 4 plus

Akshar062, 2024-12-19:
klipper
? -> v0.9.1-612

moonraker
? -> v0.7.1-0

octoeverywhere
3.6.3-6-g98dc8a

if it help

pedrolamas, 2024-12-19:
Hi @Akshar062 thank you for opening this ticket.

Please that you take a screenshot of your camera settings and paste it here.

Akshar062, 2024-12-19:
[image]
[image]

Akshar062, 2024-12-19:
[image]

this comes after refresh it is gone

[image]

pedrolamas, 2024-12-19:
Current Fluidd (which you are using) requires at least Moonraker v0.8.0-309

Your Moonraker version is far too old, either you update that to a more recent version or you will need to use whatever Fluidd version Elegoo provided.

closed 2024-12-19 by pedrolamas