fluidd-core/fluidd #1555: Add support for MPC calibration

HumbleDeer, 2024-12-27:
### Is your feature request related to a problem? Please describe

/

### Describe the solution you'd like

In addendum to commit "feat(Heaters): adds PID_CALIBRATE support" (https://github.com/fluidd-core/fluidd/commit/9682b5b)

Add support for the MPC calibration routines present in Kalico (formerly Danger Klipper).

HumbleDeer, 2024-12-27:
Thanks go out to @lukeslaboratory for the suggestion 👀

CachecticLab, 2024-12-28:
Please, make this feature optional.
It makes temperature settings barely usable.
When clicking in a temperature edit field, this action is supposed to allow the user enter a value, not run a PID calibration.
Maybe a double click or a right click, but not a single click !

pedrolamas, 2024-12-28:
> When clicking in a temperature edit field, this action is supposed to allow the user enter a value, not run a PID calibration.

Hi @CachecticLab thank you for your comment and apologies, that is actually a bug... I will fix it later today and push a new release to fix it (both left and right click will show the menu but that should not happen when clicking the target text box as it currently does)

CachecticLab, 2024-12-28:
Thanks !
Was about to revert to a previous release.
Fulidd is the best.
The one and only one that allows the user to tinker with config files. (among many other features)
Was tinkering with a new build (for a Box Turtle serial print). Was annoyed with this PID thingy. Went back to Mainsail. No way. Despite this issue, sticked with Fluidd.
Please, code a Box Turtle extension. The Box Turtle is the future.

pedrolamas, 2024-12-28:
> Please, code a Box Turtle extension. The Box Turtle is the future.

Please open a separate Feature Request for that so we can track it!

PR #1559 (2024-12-29): feat(Heaters): adds MPC_CALIBRATE support
Adds Kalico's `MCP_CALIBRATE` support.

# Option on context-menu

[image]

# Dialog

[image]

Resolves #1555
closed 2024-12-29 by pedrolamas