fluidd-core/fluidd #1556: Device copy button not working

thijstriemstra, 2024-12-28:
### Fluidd Version

1.31.3

### Browser

Firefox, Safari

### Device

Desktop PC, Mobile device (Phone, Tablet or similar)

### Operating System

Windows, iOS

### What happened

The copy buttons are not working, nothing is copied to the clipboard.

[image]

### What did you expect to happen

Content in clipboard after copy.

### How to reproduce

Copy any field and try to paste somewhere.

pedrolamas, 2024-12-28:
Hi @thijstriemstra I just tested this in Windows with Edge Dev and Firefox and it worked on both... I will perform more tests, but might be something else at play here.

Are you accessing Fluidd via http or https ?

pedrolamas, 2024-12-28:
Ok, I just tested with http only access and seems `navigator.clipboard` is not available on that context (seems it required "secure" https context).

There seems to be some workarounds, so I will look into implementing those instead.

PR #1560 (2024-12-29): fix: clipboard copy on non-secure context
The existing clipboard copy method relies on having `navigator.clipboard` access, however this is only available on "secure" https context, and always `undefined` on "non-secure" http context.

The fix here is to use an old deprecated method that seems to still work in browsers and under "non-secure" http context.

Fix #1556
closed 2024-12-29 by pedrolamas