fluidd-core/fluidd #1567: Stalling at "Connecting to moonraker..." if accessing from external location

dmit2k, 2025-01-02:
### Fluidd Version

v1.31.2

### Browser

Chrome, Firefox, Safari

### Device

Laptop, Mobile device (Phone, Tablet or similar)

### Operating System

macOS, Linux, iOS

### What happened

Just installed and updated one more printer at home, accessing Fluidd interface from the LAN works perfectly, but any attempt to connect via VPN or Nginx reverse proxy hangs at "Connecting to moonraker..." page with "Force Refresh" button.

Browser console shows no errors, all requests in fluidd-access.log and reverse proxy logs are "200". Using plain HTTP, no access control enabled. External IP address is within moonraker's enabled private IP list, but does not work even with reverse proxy installed on LAN.

Accessing my old printer (with some 6-month old Fluidd) on the same LAN via the same VPN or Reverse Proxy works perfectly.

### What did you expect to happen

To work as expected

### How to reproduce

?

pedrolamas, 2025-01-02:
Hi @dmit2k thank you for opening this ticket, however I should point out that this is not a Fluidd bug but some other network configuration issue you are experiencing.

I advise you to check your moonraker.log as it may have more information on what is going on, and if that doesn't clear the problem, jump into Klipper Discord server for more direct assistance.

closed 2025-01-02 by pedrolamas

dmit2k, 2025-01-02:
> Hi @dmit2k thank you for opening this ticket, however I should point out that this is not a Fluidd bug but some other network configuration issue you are experiencing.

Thank you very much for the prompt reply!

Unfortunately, I'm not sure that it is not a Fluidd (or its configuration) problem. I see absolutely nothing in moonraker.log while attempting to connect from outside. Looks like it gets no requests at all.

At the same time, links like 
http://10.123.123.123/webcam/?action=snapshot
http://10.123.123.123:7125/server/database/item?namespace=fluidd
are working OK. 

Would you please advise me how can I check how the Fluidd is connecting to Moonraker? It does this from my web browser or locally on the printer host?

pedrolamas, 2025-01-02:
If you don't see anything on moonraker.log, then Fluidd requests are not getting through to it.

While inside Fluidd, you can always press <kbd>F12</kbd> to open the Developer Tools and then reload the page and check the Console or Network tab for any errors.

dmit2k, 2025-01-02:
> If you don't see anything on moonraker.log, then Fluidd requests are not getting through to it.
> 
> While inside Fluidd, you can always press F12 to open the Developer Tools and then reload the page and check the Console or Network tab for any errors.

Console shows no errors, Network tab also has nothing suspicious and all requests have Status 200

At the same time and from the same device Mobileracker app is working just fine and I see connection in moonraker.log

2025-01-02 13:15:28,352 [authorization.py:_check_trusted_connection()] - Trusted Connection Detected, IP: 10.147.20.206
2025-01-02 13:15:28,354 [application.py:log_request()] - 101 GET /websocket (10.147.20.206) [_TRUSTED_USER_] 4.69ms
2025-01-02 13:15:28,355 [websockets.py:open()] - Websocket Opened: ID: 281473102685520, Proxied: True, User Agent: Dart/3.5 (dart:io), Host Name: 10.123.123.123
2025-01-02 13:15:28,368 [websockets.py:_handle_identify()] - Websocket 281473102685520 Client Identified - Name: Mobileraker-ios, Version: 2.8.3-583, Type: mobile

pedrolamas, 2025-01-02:
Try updating to the latest Fluidd release and see if the issue persists.

dmit2k, 2025-01-06:
> Try updating to the latest Fluidd release and see if the issue persists.

Just to let you know - updating Fluidd to v1.31.4 actually fixed the issue. 

Thank you for support!