fluidd-core/fluidd #1569: Support for mixing extruders

JulianStremel, 2025-01-06:
### Is your feature request related to a problem? Please describe

I am currently working on getting mixing extruder support stable in klipper and Kalico(formerly danger klipper).

### Describe the solution you'd like

Because fluid already supports multi extruder setups in its toolhead widget i wanted to ask if there would be interest in enabling a more in depth mixing extruder support from the fluid community :)
Mainly i would like to be able to set and see mixing ratios of my extruders as well as track the individual spool usage using spoolman.

### Describe alternatives you've considered

Currently i am not using any visualization and i do not have a good way to implement the spoolman feedback but one could consider building macros to set spool usage from within klipper.

### Additional information

The [original PR](https://github.com/Klipper3d/klipper/pull/5960) on the klipper repo ist stale but i am working on it over at [Kalico](https://github.com/KalicoCrew/kalico/pull/512) as i said and i would hope there is interest in the klipper community to push mixing extruders as a viable alternative to ams style systems.

I made a short Video on [my custom mixing printer](https://youtu.be/K2hD6ypBLYY?si=ieR5Ki-HbF7ZI-8L) for everybody interested

pedrolamas, 2025-02-13:
Hi @JulianStremel, thank you for submitting this Feature Request.

I see that this functionality is now merged in Kalico and I took it for a quick test, but unfortunately, I got an error:

[image]

My config has an `[extruder]` and `[extruder1]` sections, so I have no idea why this refer to a `extruder_stepper extruder1`...

This is the config I used:

```ini
[mixing_extruder]
steppers: extruder,extruder1
extruder_name: test
```

Can you help to clarify this?

JulianStremel, 2025-02-13:
@pedrolamas  Thanks for taking a look :)

your config does need to have the other extruders setup as extruder_stepper like so:

```
[extruder]
step_pin: printhead:P2.2
dir_pin: printhead:P2.6
enable_pin: !printhead:P2.1
...

[extruder_stepper extruder1]
extruder: #extruder
step_pin: printhead:P0.19
dir_pin: printhead:P0.20
enable_pin: !printhead:P2.8
microsteps: 16
rotation_distance: 23.5

[extruder_stepper extruder2]
extruder: #extruder
step_pin: printhead:P0.22
dir_pin: printhead:P2.11
enable_pin: !printhead:P0.21
microsteps: 16
rotation_distance: 23.5

[extruder_stepper extruder3]
extruder: #extruder
step_pin: printhead:P2.13
dir_pin: printhead:P0.11
enable_pin: !printhead:P2.12
microsteps: 16
rotation_distance: 23.5

[mixing_extruder]
steppers: extruder,extruder1,extruder2,extruder3
```
I have a little repo for some of my configs:
https://github.com/JulianStremel/Ender3-mixing

Also switching the mixing ratio mid print is currently only possible if 

`
#define CHECK_LINES 1
`
gets set to:
`
#define CHECK_LINES 0
`
in [klippy/chelper/stepcompress.c](https://github.com/Klipper3d/klipper/blob/fec3e685c92ef263a829a73510c74245d7772c03/klippy/chelper/stepcompress.c#L28)

As far as i can tell this does not break anything else and is needed because this check as far as i can tell checks if the send steps are okay in regard to the rotation distance but because the mixing feature builds on dynamically changing the rotation distance for every stepper that is part of that extruder this check breaks.

There are some more quirks like having to apply pressure advance to every extruder_stepper seperately and not just to the parent extruder... but i think this should get you started.

pedrolamas, 2025-02-13:
Thank you for the information, I will check on that ASAP.

Just for clearing, Klipper and Kalico allow and work fine with `[extruder]`, `[extruder1]`, I assume this is not supported then?

JulianStremel, 2025-02-13:
Klipper supports the standard way of defining multiple extruders for toolhead setups and stuff like that and kalico beeing a fork also supports that.

The only thing i added was the support for a hotend that shares multiple filament extruders that feed into a single meltzone for mixing applications.