fluidd-core/fluidd #1571: Utilize vertical screen space for Jobs and Configuration screens.

charredchar, 2025-01-09:
### Is your feature request related to a problem? Please describe

There is a lot of dead space on the Jobs and Configuration screens with a high enough resolution. This can make looking through the menus on these pages more difficult. This is in contrast to screens like the Console which will resize the elements to fit the window.

### Describe the solution you'd like

Have the elements within these screens use all available space in the window.

### Additional information

I personally keep the Fluidd UI tab open on a portrait screen which gives me more vertical space. This is fantastic for the console, for example, but with the Jobs and Configuration screen the elements stay short and scrolling becomes necessary even when there is enough space for the lists in those elements. This is also noticeable on a higher resolution landscape screen.
An example of what I expect;
[image]

What I get instead;
[image]
[image]

pedrolamas, 2025-02-10:
Hi @charredchar, thank you for submitting this feature request.

I've now started working on this and have it ready for testing!

Having said that, I have 2 versions to try out:

1. Set the card `height` to the available useful space (just like the Console view currently does)
2. Set the card max-height, which means the card will be allowed to expand till the available useful space

I have no real preference, though I am inclined to version 2 as it keeps the user focus...

closed 2025-02-11 by pedrolamas

charredchar, 2025-02-14:
> Hi [@charredchar](https://github.com/charredchar), thank you for submitting this feature request.
>
> I've now started working on this and have it ready for testing!
>
> Having said that, I have 2 [...]

Tha is for your work!
If version 2 was to be used would there be a way to always ensure it used all available space like version 1? Maybe a tick option next to the `max-height` field to toggle it to disable the field and always expand to full height like version 1?
The only issue I could see with version 1 is mobile device view. Though I really never use my phone for Fluidd so I'd have no input on that.

PR #1591 (2025-02-11): style: layout changes to allow full height
Made the content in pages Configuration, Jobs, Console and Timelapse use the full vertical available space.

Also, I hate layout work...

Resolves #1571