fluidd-core/fluidd #1575: after i do a bed mesh it save save to config but the button doesnt light up for me to click on it

vinny-svg, 2025-01-16:
### Fluidd Version

v1.31.4

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

the save and config button doesn't light up 

### What did you expect to happen

it to light up so i can save and start printing 

### How to reproduce

i went back to my stock printer setting

pedrolamas, 2025-01-16:
Hi @vinny-svg, thank you for opening this ticket.

What do you mean with "save and config button doesn't light up"? That you don't see this button showing up?

[image]

Mind you, if this is the case, you do need to enable it on Fluidd Settings:

[image]

vinny-svg, 2025-01-16:
What I mean is it's there but not like the pic it's a light grey color and u can't click on it and it is set up in the settings as well  [image]

pedrolamas, 2025-01-16:
It should be disabled if you have a print in progress (so Klipper is reporting "printing" or "paused" state).

Is your printer idle?

vinny-svg, 2025-01-16:
It was at idle then it did the bed mesh then wanted it to be saved and re start but after it was done I couldn't never was able to even do a print

vinny-svg, 2025-01-16:
I can get a picture later I'm not neat it right this second and it's a k1c and I know its v1.31.4 but as soon as I can I'll get a picture

vinny-svg, 2025-01-16:
[image]

pedrolamas, 2025-01-16:
I can see from the screenshot you are using the latest Fluidd and Moonraker, but given you have a Creality K1 you are using a fork of Klipper, is that so?

We only support mainline Klipper and Kalico, anything else is out of scope.

vinny-svg, 2025-01-16:
Not sure I never had this issue tell the update came out

pedrolamas, 2025-01-16:
We only test with mainline Klipper and Kalico, I have no idea what changes you have in your local Klipper install which most likely are causing this problem.

I'll be closing this ticket now, but if you update to mainline Klipper and still have this issue I'll be sure to reopen it again.

closed 2025-01-16 by pedrolamas

vinny-svg, 2025-01-16:
How do I go about updating to it then I don't understand how for months it's fine tells simply update but thanks for getting back to me.