fluidd-core/fluidd #1578: Adjust layout drag/drop not working OrcaSlicer macOS

thijstriemstra, 2025-01-24:
### Fluidd Version

1.31.4

### Browser

Other (Please tell us below)

### Device

Laptop

### Operating System

macOS

### What happened

Trying to drag and drop items in the layout adjust screen isn't working. It's possible to drag the items but dropping doesn't work.

### What did you expect to happen

Able to re-arrange items.

### How to reproduce

Try OrcaSlicer 2.2.0 on macOS 12.7.6 with fluidd.

### Additional information

I'm not sure what backend browser OrcaSlicer uses on macOS, it's MS Edge on Windows where drag/drop works ok. Also tested it in local Safari browser and seems to work fine.

pedrolamas, 2025-01-25:
Hi @thijstriemstra, thank you for opening this issue, however it will depend on Orca to fix this problem as we consider this scenario unsupported.

For what is worth, I can see that drag & drop works fine outside of Orca in MacOS.

closed 2025-01-25 by pedrolamas

thijstriemstra, 2025-01-25:
> For what is worth, I can see that drag & drop works fine outside of Orca in MacOS.

Ok, I'll adjust the dashboard in Safari (which also changes it in OrcaSlicer it seems).