fluidd-core/fluidd #1581: Can't set [temperature_fan] anymore after updating to 1.32.0 on one printer, but I can on another

Atomique13, 2025-02-08:
### Fluidd Version

1.32.0

### Browser

Chrome

### Device

Desktop PC

### Operating System

Windows

### What happened

I updated from 1.31.4 to 1.32.0 and now I get the error you can see in the attached screenshot. It was working as intended on 1.31.4, I double checked this by rolling back fluidd to the old version and it works with no error, same cfg.

Updated to 1.32.0 on another printer and that one has no problem setting a [temperature_fan] value.

[image]

### What did you expect to happen

To be able to set temperature_fan.

### How to reproduce

.

### Additional information

1st printer cfg that doesn't work anymore with new fluidd update but works with 1.31.4

```
[temperature_fan HOST_fan]
pin: PA1
kick_start_time: 0.5
max_power: 1.0
min_temp: 0
max_temp: 90
hardware_pwm: true
target_temp: 60
sensor_type: temperature_host # please comment out or remove the [temperature_sensor Host_temp] section if you don't want the temperature shown twice in the interface.
max_speed: 1.0
min_speed: 0.0
control: pid
pid_Kp: 2.0     ;40
pid_Ki: 5.0     ;0.2
pid_Kd: 0.5     ;0.1
pid_deriv_time: 2.0
```

2nd printer cfg that works with new fluidd update
```
[temperature_fan soc_fan]
pin:PB2
sensor_type: temperature_mcu
control:watermark
kick_start_time:0.500
shutdown_speed:0
off_below:0.1
target_temp:45
min_temp:-15
max_temp:100
max_speed:1
min_speed:0.3
tachometer_pin:PC6
```

pedrolamas, 2025-02-08:
Hi @Atomique13, thank you for reporting this issue.

I have confirmed the issue and am working on a fix that I will release ASAP.

closed 2025-02-08 by pedrolamas

Atomique13, 2025-02-08:
Updated to latest on both machines, it works as intended now. Thank you for the fast fix. Wish you a great day!

PR #1582 (2025-02-08): fix: printer settings must use lowercase key
This bug was introduced with the changes on 897e163bb9b39b91cd927e90c63cf77de53b0dfd, as printer settings are all lowercase keys however this was not ensured at the time of refactor.

Fixed #1581