fluidd-core/fluidd #1603: verbose gcode from orcaslicer breaks gcode preview

shackleford92, 2025-02-16:
### Fluidd Version

1.32.2

### Browser

Firefox

### Device

Laptop

### Operating System

Windows

### What happened

i recently went through a few days wondering why my gcode preview was blank.
i isolated and rectified the issue, i had enabled verbose gcode in orcaslicer.
not sure if this is a limitation or a bug, but the parsing of comments seems to gum up gcode preview.
result is no ability to use gcode preview, no load file, no clear file, just blank disabled for the duration of the print.

### What did you expect to happen

i dont know if i should expect comment parsing, but i thought it would continue to work with verbose gcode enabled

### How to reproduce

use up to date orcaslicer as of 02/15/2025, enable verbose gcode in other settings>gcode output>verbose gcode, slice, upload and print, look at gcode preview. 

### Additional information

its also broken in orcaslicer, which seems to be using edge, which is chromium. i say verbose gcode, but its just comments, unsure if this applies to all comments, possibly? not the expected linearity due to the existence of lines that are just comments? my ignorant speculations

pedrolamas, 2025-02-16:
Hi @shackleford92, thank you for opening this issue.

I have however not been able to reproduce the issue with Fluidd v1.32.2, all my files sliced in Orca with "verbose gcode" enabled are loading fine in the gcode preview.

Can I ask that you upload a sample gcode file where you are experiencing this so I can take a look please?

shackleford92, 2025-02-16:
here you gp

[3DBenchy_T=46m6s_W=13g_C=$0.28.zip](https://github.com/user-attachments/files/18815410/3DBenchy_T.46m6s_W.13g_C.0.28.zip)

pedrolamas, 2025-02-16:
Thank you @shackleford92, I can manually load your file just fine!

[image]

> correction, it works only if i load the preview from jobs during the print, the auto load seems to be failing for me

are you saying that it is the automatic load that is not working?

pedrolamas, 2025-02-16:
From my tests with your file just now, Fluidd v1.32.2 is auto-loading the gcode file on print start without any problems, so I am still not able to reproduce the issue here.

Are there any other steps you are performing?

shackleford92, 2025-02-16:
im running a 3rd party firmware on a cheap printer, it was a nice working feature when it worked
but no, fusion360/thingiverse > import stl to orca > slice print

gcode preview on homepage is initially greyed out
i can manually open the file from jobs into gcode preview, from where it acts as normal and will follow progress properly. 
i can also oddly seem to load jobs which arent printing which it also tries to follow, idk how its following anything but layer height but i guess some extrusions overlap?

pedrolamas, 2025-02-16:
Can you confirm what version Fluidd shows on the bottom right of the page? Official version should show `1.32.2-d7e03cf` (just like my screenshot above)

If you are using a fork of Klipper, Moonraker or Fluidd, then there is little we can do here to help you with as we won't have visibility to any changed done on said forks.

shackleford92, 2025-02-16:
1.32.2.d7e03cf correct
ive been getting pushed a bunch of updates for fluidd lately

pedrolamas, 2025-02-17:
> would it be relevant that the files not auto loading are on a thumb drive in a very overloaded usb bus, split into a hub, doing wifi cam and the thumb drive?

Maybe?

You might want to press F12 to bring up the browser Developer Tools, change to the Console tab, and check if any error is logged there... if so, please take a screenshot and paste it here!

PR #1607 (2025-02-18): fix: get directory for current printing file
For a long time now, we force the load of the metadata for the current printing file, and that works fine.

However, the way that the files store works, we store files and folders information on an object with root path as key - and loading the metadata for a file whose folder does not exist on the object would just discard said data!

To fix this, we will now ensure we load the full file parent directory (which will include the metadata for all files) instead of just the metadata for this specific file.

Fixes #1603 and #1606
closed 2025-02-18 by pedrolamas