fluidd-core/fluidd #1604: PROBE_ACCURACY not available for BTT Eddy

thijstriemstra, 2025-02-16:
### Fluidd Version

1.32.2

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

The PROBE_ACCURACY command is not available when using BTT Eddy probe:

[image]

 

### What did you expect to happen

That the command is available for `[probe_eddy_current]` devices.

### How to reproduce

Define an `[probe_eddy_current btt_eddy]`

pedrolamas, 2025-02-16:
Hi @thijstriemstra, thank you for opening this issue.

I've changed the status from "bug" to "feature request" as we never added support for `probe_eddy_current` objects, but we will add it in time for next release!

pedrolamas, 2025-02-16:
For what is worth, reading through the Klipper documentation tells me that `PROBE_ACCURACY` is not just for BTT Eddy but for any probe in the system, but we currently do not show that command in the interface (only the `PROBE_CALIBRATE`)

The main issue I see is if we start adding all of these we quickly can run out of space!

thijstriemstra, 2025-02-16:
understood, but since fluidd is a frontend that makes things easier: i keep typing something (and having to remember syntax) that should be as easy as a button click imo. it also helps people that are new and perhaps didn't even know the PROBE_ACCURACY command is available. oh well

pedrolamas, 2025-02-19:
> Maybe it's time for a diagnostics mode for probes where only these "advanced" options are visible..

I do like that idea! No promises, but I will be looking at this when I get the chance! 🙂

thijstriemstra, 2025-02-19:
i think it makes sense to move all probe-related tuning and calibration options into the "Tune" panel, just like bed mesh calibration being located there. z-tilt and other calibration tools could live there as well?

thijstriemstra, 2025-04-19:
I think this can be closed, it's working now.

closed 2025-04-19 by thijstriemstra