fluidd-core/fluidd #1616: Issue with spool QR-code detection

jaarenhoevel, 2025-02-26:
### Fluidd Version

v1.32.3-9eb0e37

### Browser

Firefox, Chrome

### Device

Desktop PC

### Operating System

Linux

### What happened

Spool detection only works when QR-code is already in view when opening "Scan Spool" dialog.

https://github.com/user-attachments/assets/b85325d2-57dc-43cc-8075-56ae2ae0c1e1

### What did you expect to happen

Continuous scanning for QR-codes while dialog is open.

### How to reproduce

- Open "Scan Spool" dialog with no QR-code visible and then move it into view.
- Close and reopen dialog while leaving the QR-code in view.

### Additional information

Tried with MJPEG Stream and MJPEG Adaptive.

Sometimes switching browser tab back and forth also works.

Camera resolution is 2048x1536px, provided by crowsnest.

closed 2025-03-03 by pedrolamas

jaarenhoevel, 2025-03-04:
Thanks for looking into this! The update fixed the spool detection with MJPEG Adaptive.

Not a problem for me, but just wanted to add that it still seems to be broken with MJPEG Stream cameras. If you need more information on my setup please let me know.

pedrolamas, 2025-03-04:
Hi @jaarenhoevel, thank you for providing an update on this!

> Not a problem for me, but just wanted to add that it still seems to be broken with MJPEG Stream cameras. If you need more information on my setup please let me know.

I did test it with both MJPEG Adaptive and MJPEG Stream on my environment and both worked correctly with the fix, but I will make some more tests to make sure I didn't forget anything!

PR #1621 (2025-03-03): fix(Cameras): always emit frame event
Ensures that frame event is always raised by handling it directly on the camera mixin (and allowing to override this on the implementations)

Fixes #1616