fluidd-core/fluidd #1618: Estimated Finish time may be displaying 'pm' instead of 'am'

dopreo, 2025-03-02:
### Fluidd Version

v1.19.0-6184c73

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

In the Fluidd UI, a file is forecasted to be around **10 hours** in total print time.

As one can see in the following screenshot, the duration thus far has been around 5 hours:

[image]
_It was around **19:37** (.53s) in my timezone (GMT, at parity with UTC) when I took this screenshot._

The file is estimated to be **about 10 hours in total duration**, and the file **started around 5 hours ago**.
This suggests, then, that the file could thus complete **another 5 hours or so from now**, taking us to around **02:30** (**2:30am**).

However, Fluidd's UI states an estimated finish time of **02:29pm** (twelve hours ahead of the reasonable estimate). This is **more like 17 hours** away.
This leads me to believe that Fluidd may be **accidentally showing** '**pm**' next to my time, instead of '**am**'.

For this time to make sense, Fluidd's timezone would need to be about **twelve hours _after_** or **twelve hours _before_** UTC. It is not.
As such, I'm led to believe that Fluidd may be inadvertently displaying this time as 'pm' rather than 'am'.

### What did you expect to happen

I would expect for this time to **instead be displayed as** '**am**', in accordance with the rest of the estimates (from / with which it is presumably being calculated).

I would find it very strange that the finishing time estimate calculated would truthfully be **twelve hours in front** of what the other numbers would suggest.

### How to reproduce

I've yet to confirm this, but:
- Start a longer print that is to **conclude the day after it is started** (according to the timezone Fluidd is using)
- View the **estimated finishing time** in the Printing dialogue box on the Home menu
- It may display '**pm**' when displaying an '**am**' time

### Additional information

**I apologise if I've left this in the wrong place**; I mean only to point this out in case it hasn't been noticed before.

I used Orca Slicer for this print.
To my incredibly limited knowledge, it does not pass on information about timings such that it would affect this, though I may very well be mistaken.

pedrolamas, 2025-03-02:
Hi @dopreo , the version you are using is more than 2 years old.

I recommend that you update to the latest (v1.32.3 at the time of writing) as that issue is not present there.

closed 2025-03-02 by pedrolamas

dopreo, 2025-03-02:
Hi there,

I am so incredibly sorry to have brought this up unbeknownst to this. This is something I should have checked beforehand more thoroughly; I was under the impression I had updated to a more recent version, but very clearly not.

Thank you for your swift response, and I am glad to hear the issue is not present in future versions. I find it very typical that the version from the factory is so old.

Thanks for your time!