fluidd-core/fluidd #1624: Can't start print after update

Meg1PrinterBackup, 2025-03-04:
### Fluidd Version

1.32.4

### Browser

Chrome

### Device

Laptop

### Operating System

Windows

### What happened

after updateing my qidi plus 4 to the newest version of fluidd print jobs can not be started

### What did you expect to happen

start printing

### How to reproduce

update to the actual version of fluidd

### Additional information

reinstalling V1.32.3 printer works as expected

pedrolamas, 2025-03-04:
Hi @Meg1PrinterBackup, thank you for reporting this issue.

Can you please be a bit more descriptive on what you are experiencing?

Also, by any chance are you using Spoolman? If so, your issue will be related with #1622 which I fixed a few hours ago and will release a fixed version later today.

Meg1PrinterBackup, 2025-03-04:
yep - I'm using spoolman - ok, I'll be patient...