fluidd-core/fluidd #1628: Humidity data is not displayed when the sensor name contains uppercase letters

VictorStatko, 2025-03-06:
### Fluidd Version

v1.32.6-98f1ebb

### Browser

Chrome

### Device

Desktop PC

### Operating System

Linux

### What happened

When using the SHT3x sensor, the relative humidity reading is only displayed if the sensor name is written in lowercase. If the sensor name contains any uppercase symbols, the humidity data is hidden. This issue may also occur with other sensors, but it has only been confirmed with the SHT3x so far.

[image]
[image]
[image]
[image]

### What did you expect to happen

The relative humidity data should always be displayed (if available) and should not depend on the sensor name's casing (uppercase or lowercase).

### How to reproduce

Test different sensor names, including those with and without uppercase letters, to observe the behavior.

pedrolamas, 2025-03-06:
Hi @VictorStatko, thank you for reporting this issue.

I have however not been able yet to reproduce this issue... can I ask that you please upload your klippy.log so I can take a look and see if I can find anything that will help me explain this?

VictorStatko, 2025-03-06:
Sure 

[klippy.log](https://github.com/user-attachments/files/19110486/klippy.log)

VictorStatko, 2025-03-06:
@pedrolamas FYI, this issue has been reproduced and confirmed by multiple users (iDryer Unit filament dryer owners). It appears that all of them have a similar configuration.

Arksine, 2025-03-06:
FWIW, there is some discussion of this on discourse.

https://klipper.discourse.group/t/lost-display-of-bme680-sensor-data-in-fluidd-plus-a-puzzling-http-404-not-found-error/22341/16

The 404 has been corrected.  I did find an instance where something is sending an invalid subscription request, however I'm not sure its Fluidd as there are multiple clients making requests at the time of the log entry.

PR #1631 (2025-03-06): fix(Thermals): show extra sensor data
This issue was introduced with the changes on 897e163bb9b39b91cd927e90c63cf77de53b0dfd, settings keys are lowercase and this was missing here!

Fixes #1628
closed 2025-03-06 by pedrolamas