fluidd-core/fluidd #1639: Display latest klipper version in system mcu sections

thijstriemstra, 2025-03-11:
### Is your feature request related to a problem? Please describe

I'd like to know how outdated my mcu 's are  without having to navigate away from the page and lookup the installed klipper version. Version nrs I check are marked in yellow in picutre below:

[image]

### Describe the solution you'd like

Show installed klipper version next to the mcu section version nr or anywhere on the page to compare to.

### Additional information

fluidd 1.32.6

pedrolamas, 2025-04-20:
Hi @thijstriemstra, thank you for submitting this Feature Request.

Adding that information should be relatively trivial (could be just an icon with a tooltip for the runtime Klippy version) but AFAIK, Klipper will show a warning if an MCU needs to be updated, so I'm not sure of what value one takes from having this indication here!

thijstriemstra, 2025-06-09:
Fair enough.