fluidd-core/fluidd #1643: Gcode preview zoom doesn't account for a wide brim

yoyo42, 2025-03-15:
### Fluidd Version

v1.3.0.4-1811cada

### Browser

Other (Please tell us below)

### Device

Desktop PC

### Operating System

Windows

### What happened

I used Orcaslicer 2.2.0 to slice a model and added a 10mm wide brim to the objects. When loaded into the Gcode Preview on the Device tab that Orca brings up when printing then if I use the Zoom button half of the brim is cut off around the edges of the grid area. (The Device tab is just Fluidd on an embedded browser window but I don't know which browser it uses for that).

### What did you expect to happen

I expected the whole object including brims to be on screen after the autozoom.

### How to reproduce

1. create/load any part onto the build plate in Orcaslicer
2. Set a wide brim
3. slice the model
4. upload the model
5. view it in the Fluidd gcode preview window
6. use the autozoom button

### Additional information

[image]

pedrolamas, 2025-03-15:
Hi @yoyo42, thank you for opening this ticket.

I should however point out that this is actually by design: the zooming functionality is to printed object which excludes layer 0 as it can contain purge lines or brims.

Having said that, I guess we can add a configurable option to allow users to specify if they want to follow this behavior or zoom the whole used printed area.

yoyo42, 2025-03-15:
That makes sense, thanks. Since it's easy to manually zoom out a little and is definitely by design then I'm happy to close this again. Thanks for the quick response!