fluidd-core/fluidd #1651: go2rtc webcams don't work in Firefox

WayneManion, 2025-03-21:
### Fluidd Version

1.30.6

### Browser

Firefox

### Device

Desktop PC, Laptop, Mobile device (Phone, Tablet or similar)

### Operating System

Android, Linux

### What happened

I have some printers. Some of them have integrated cameras. All of these types of cameras are exposed as mjpeg. Examples: Creality K1 and Sovol SV08. 

I also have some older printers without cameras. Examples: Sovol SV06, Ender 3 Pro. I have PoE IP webcams pointed at these printers. The PoE RTSP feed goes into go2rtc to be consumed by Frigate and by fluidd. 

I use moonraker.conf to define the camera for each printer. This way the camera is already set up and working when I view the fluidd UI in a different browser or on a different machine. 

If I open fluidd interface for any of my printers in Chrome or Brave, the stream works fine inside of fluidd. 

[image]

If I open fluidd in Firefox, the printers with mjpeg streams work fine, but the go2rtc streams do not work. This only happens in Firefox. It doesn't seem to matter if I use moonraker.conf to configure the stream or if I add the stream through fluidd. If I use the hls stream provided by go2rtc, this works. 

[image]

### What did you expect to happen

I wish for my go2rtc cameras to appear in fluidd on Firefox when configured via moonraker.conf or via fluidd UI. 

### How to reproduce

Create a go2rtc stream. Add the stream in fluidd UI or in moonraker.conf. Watch the stream appear as normal in fluidd on Chrome. Do the same thing in Firefox and watch the camera not appear. 

### Additional information

My camera is defined in moonraker.conf as such:
```
[webcam silvio]
location: printer
enabled: True
service: webrtc-go2rtc
target_fps: 15
target_fps_idle: 5
stream_url: http://192.168.1.32:1984/webrtc.html?src=silvio&media=video+audio
snapshot_url: http://192.168.1.32:1984/api/frame.jpeg?src=silvio
flip_horizontal: False
flip_vertical: False
rotation: 0
aspect_ratio: 16:9
```

pedrolamas, 2025-03-21:
Hi @WayneManion thank you for opening this ticket.

Can you please confirm what is full Fluidd version you are using, as it is shown on the bottom right corner of every page?

pedrolamas, 2025-03-21:
Hi @WayneManion, I managed to run a few tests and it seems that the issue is upstream in the [go2rtc reference player implementation](https://github.com/AlexxIT/go2rtc/blob/master/www/video-rtc.js) that we use in Fluidd (same goes for Mainsail) and somehow is failing.

Their own WebRTC players are failing on Firefox, so we will need them to check and fix this before we can do anything from our side.

WayneManion, 2025-03-24:
I don't see the version number on every page, but on the software update part of the settings page I see:

 fluidd
v1.32.6

pedrolamas, 2025-03-24:
> I don't see the version number on every page, but on the software update part of the settings page I see:

Please scroll the page all the way to the very bottom, you should then see something like this:

[image]

If you don't see something like the above, then I think you might not be using mainline Fluidd but some fork.

WayneManion, 2025-03-25:
© 2025 fluidd: v1.32.3-9eb0e37

pedrolamas, 2025-03-27:
I've been intrigued with this issue and so have been doing some research on it, turns out that if go2rtc is running under the same host as the web page in which we are playing the stream (in this case, Fluidd), it will not work!

However, if I run go2rtc on a separate host and then set the camera address in Fluidd to that other host, it works perfectly!

The only way I managed to get it to work with both Fluidd and the go2rtc serving from the same host was by going to "about:flags" in Firefox and forcing `media.peerconnection.ice.obfuscate_host_addresses` to `false`, but that can create some security issues so I would not recommend this action.

WayneManion, 2025-03-27:
That is not my experience. I have multiple machines running Klipper. One is an x86 PC. Two are CB2s mounted on BTT Manta boards. go2rtc is running on a completely separate machine that also runs Frigate. go2rtc streams running on the Frigate box do not work when trying to use them as the stream source for any of the Klipper machines.

Clarification:
I am using my desktop PC. I have a Firefox browser window open using fluidd on the 3D printer, a Creality K1 with a Manta M5P and CB2 inside. The Frigate machine is running go2rtc (and Frigate). 

I can use and see the mjpeg feed coming from Crowsnest on the K1 in fluidd on Firefox on the desktop PC.
I can see and use the hls feed coming from the Frigate box in fluidd on Firefox on the desktop PC. 
I cannot see or use the rtc feed coming from the Frigate box in fluidd on Firefox on the desktop PC. It does work in Chrome though.

pedrolamas, 2025-03-27:
Thank you for that clarification @WayneManion, can I ask that you run a simple test in Firefox:

- opening `about:config` (click on "proceed" when shown the warning)
- enter `media.peerconnection.ice.obfuscate_host_addresses` on the textbox
- change the value to `false` by clicking on the "alternate" button on the right

[image]

With that value set to `true`, open Fluidd and see if the WebRTC (go2RTC) stream now shows correctly.

If it doesn't, please press F12 to open the Developer Tools, change to the console tab, and if you see any error there please take a screenshot and post it here.

After that, you can revert the change above by setting that value back to `false`.

pedrolamas, 2025-03-27:
> [stream.html](http://frigatehost.local:1984/stream.html?src=sven&mode=webrtc) WebRTC stream / browsers: all / codecs: H264, PCMU, PCMA, OPUS / +H265 in Safari

Does opening that link directly on Firefox work for you?

pedrolamas, 2025-03-28:
I've done a few more tests by using Frigate myself and pointing it to one of my cameras; I then opened the go2rtc links page on the browser (matched the screenshot you posted above) and quickly realized that I could only use the topmost link of the "Any codec in source" list... neither the topmost link on the "H264/H265 source" or the "MJPEG source" would play on my Edge browser!

Frigate is using go2rtc v1.9.2, which is relatively old and probably has some bugs that have since been fixed.

If I use the latest go2rtc (so bypass frigate completely) it works fine on Edge browser, and also on Firefox **but only after I make that setting change discussed above!**

WayneManion, 2025-03-28:
I started up a go2rtc docker image with the latest 1.9.9 version. 

I think the top link (Any source in codec) just uses a different stream if rtc doesn't work. In my case, when I click the Any source in codec link in Firefox it uses the MSE stream. You can see MSE in the top-right corner of the browser window. Brave does the same thing. 

When you follow the (Any source or codec) link I don't think you're actually seeing an RTC stream. 

When I click the "[stream.html](http://klipperbox.arpa:1984/stream.html?src=sven&mode=webrtc) WebRTC stream / browsers: all / codecs: H264, PCMU, PCMA, OPUS / +H265 in Safari" link in Firefox the stream will not play. It says: 

[image]

pedrolamas, 2025-03-28:
From my perspective, that settles the issue: Fluidd supports go2rtc WebRTC streams only; if these don't work with their own player in Firefox, there is nothing we can do on our end until they fix in from their side first.

For the record, I did manage to get it working with go2rtc latest version (no Frigate) and by changing that key in Firefox.

closed 2025-03-28 by pedrolamas