fluidd-core/fluidd #1653: Incorrect disk usage calculation

thijstriemstra, 2025-03-26:
### Fluidd Version

1.33

### Browser

Firefox

### Device

Desktop PC

### Operating System

Windows

### What happened

After installing a very common raspberry pi tool called [log2ram](https://github.com/azlux/log2ram) I've been getting incorrect disk usage stats, including a new icon in the file explorer:

[image]

[image]

```console
$ sudo lsblk -o NAME,FSTYPE,SIZE,MOUNTPOINT
NAME   FSTYPE   SIZE MOUNTPOINT
sda           117.2G
├─sda1 vfat     512M /boot/firmware
└─sda2 ext4   116.7G /
```

```
 sudo fdisk -l
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 4096 bytes
I/O size (minimum/optimal): 4096 bytes / 4096 bytes

Disk /dev/sda: 117.19 GiB, 125829120000 bytes, 245760000 sectors
Disk model: PHILIPS
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xe50164e2

Device     Boot   Start       End   Sectors   Size Id Type
/dev/sda1          8192   1056767   1048576   512M  c W95 FAT32 (LBA)
/dev/sda2       1056768 245759999 244703232 116.7G 83 Linux
```

### What did you expect to happen

Show correct disk stats.

### How to reproduce

not sure if log2ram is the cause.

thijstriemstra, 2025-03-26:
it's also hit or miss: sometimes it shows the correct disk stats, sometimes it doesn't, totally random it seems.

pedrolamas, 2025-03-26:
Hi @thijstriemstra, thank you for reporting this issue.

I recognize this behavior is not expected or ideal, however, Fluidd has no control over this data as it is provided by the Moonraker API.

If there is a bug on this area, it will have to be fixed there.

thijstriemstra, 2025-03-27:
Thanks for the feedback, I tried finding the issue by looking throught the moonraker code but nothing obvious stood out, maybe @Arksine has any idea?

Arksine, 2025-03-27:
Moonraker uses Python's [shutil](https://docs.python.org/3/library/shutil.html#shutil.disk_usage) module to report disk usage.   I'm not sure why installing log2ram would affect it, but based on the screenshots it seems Fluidd reporting usage from the tmpfs mount.

It might be worth enabling verbose logging in Moonraker and reproducing.   With that we can inspect the result from a `directory` query on the `config` folder and see if its reporting the wrong data.

thijstriemstra, 2025-03-29:
> With that we can inspect the result from a directory query on the config folder and see if its reporting the wrong data.

I enabled verbose logging but any idea what grep command I'd use on moonraker.log to find the `directory` query?

Arksine, 2025-03-31:
That looks like the correct response.  Its reporting ~14GB total, with ~5GB free.  I suspect that the most likely cause is that Fluidd expects all of the root folders ("gcodes", "config", "logs", etc) to be on the same partition, and Fluidd expects that the disk will only have one primary partition.

pedrolamas, 2025-03-31:
> I suspect that the most likely cause is that Fluidd expects all of the root folders ("gcodes", "config", "logs", etc) to be on the same partition, and Fluidd expects that the disk will only have one primary partition.

Correct on that one: though this is legacy code, I guess we never had a scenario where multiple disks to be in use!

Not sure what is the best approach to "fix" this one...

pedrolamas, 2025-03-31:
Ok, might have to "kill" this one and replace with an inline indicator on each of the file system cards instead...

Bottom line is that the bug is correct and a Fluidd one for that fact, so I am reopening this ticket.

Arksine, 2025-03-31:
Given time I could probably implement a `machine` endpoint that provides general disk and partition info.  Alternatively it should be possible to report the partition name in the `disk_usage` field of the directory response.  I may be able to get that done more quickly as I don't have to research how to pull all the useful disk info for block devices.

In the interim Fluidd could associate capacity info with "roots", perhaps provide a dropdown or way to select a root

thijstriemstra, 2025-03-31:
thanks for the feedback @Arksine. can anyone explain the randomness though?

> it's also hit or miss: sometimes it shows the correct disk stats, sometimes it doesn't, totally random it seems.

pedrolamas, 2025-03-31:
I can explain that one: we take the risk space info from the first loaded response! And that is a bit random as any of the roots can return first...

It's definitely a bug and we will fix it (most likely by removing that card and adding the space information on each root file explorer)

closed 2025-04-01 by pedrolamas

thijstriemstra, 2025-04-09:
I still see a low disk warning for the logs:

[image]

even though 1% of the disk is full:

[image]

pedrolamas, 2025-04-09:
@thijstriemstra can you check if you have any error on the browser Console in the browser developer tools?

thijstriemstra, 2025-04-09:
I don't see any..

pedrolamas, 2025-04-09:
Ah, wait a second, this is actually correct as currently we assume 1Gb as the minimum to show that alert (regardless of root directory)!

Having said that, it might make sense to have these limits per root directory as logs don't actually need 1Gb (maybe 100Mb instead?)

thijstriemstra, 2025-04-09:
> Having said that, it might make sense to have these limits per root directory as logs don't actually need 1Gb (maybe 100Mb instead?)

Do you have access to total size? I would say 80% of total disk size triggers a warning?

pedrolamas, 2025-04-09:
I like your way of thinking: might do 80% or 1Gb, whichever is smaller, and call it a day!

pedrolamas, 2025-04-09:
> I like your way of thinking: might do 80% or 1Gb, whichever is smaller, and call it a day!

Implemented on 285278f8f59a22ab9d999409339a8e03384ff7be

thijstriemstra, 2025-04-09:
thanks for your awesome work @pedrolamas, as always. Would love to test it without compiling myself. Would a nightly build thing be an option for fluidd?

pedrolamas, 2025-04-09:
Every commit generates a build, so you can download it from the GitHub actions for the commit above.

Alternatively, you can use Docker to run the latest image generated from the develop branch (this is what I do myself!)

thijstriemstra, 2025-04-12:
also; would be nice when clicking the low disk warning icon it takes you to the relevant system disk usage section. Right now clicking the button doesn't do anything.

PR #1660 (2025-04-01): feat: show disk info per registered directory
This splits part of the existing Disk Information card to a new card, SD Card Information.

On the existing Disk Information card we now show tabs for each registered directory root so we can show disk space info for each one of them!

## SD Card Information and Disk Information cards

[image]

Fixes #1653