fluidd-core/fluidd #1662: MMU: Check gates

SHKinsem, 2025-04-04:
### Is your feature request related to a problem? Please describe

When using ERCF v1.0, it appears to be annoying to type the macro to check if a gate is loaded with filament (after this gate is marked as empty)

### Describe the solution you'd like

Can we add a series of buttons just like the tool change, but with function of checking if a certain gate is loaded with filament

moggieuk, 2025-04-04:
Good idea, but it sounds like you may no be using the new UI....?

Before I complicate the UI though, are you aware that you can check the SELECTED  gate with a button already?  

[image]

Note that you can also use the "Recovery dialog" to directly tell Happy Hare the gate state.

Your idea might be a good one -- the "gate status icon" (the numbered panel) part of the spool display could double as a button to initiate this action.... but before I complicate the UI I want to make sure what is there isn't sufficient

SHKinsem, 2025-04-06:
Somehow I did not see the ui in your screenshot

[image]

Here is what I have now, using fluidd v1.33.0
Is there a guide to enable these functions?

SHKinsem, 2025-04-06:
OK I just updated to v1.34.0, and now I can see the ui
I will try it for a few days and give feedback

moggieuk, 2025-04-09:
Cool!

closed 2025-05-18 by SHKinsem