fluidd-core/fluidd #1671: missing thumbnail on 1024x600 touchscreen

thijstriemstra, 2025-04-10:
### Fluidd Version

1.34.1

### Browser

Chrome, Other (Please tell us below)

### Device

Mobile device (Phone, Tablet or similar)

### Operating System

Linux

### What happened

the gcode image is missing on a 1024x600 fullscreen chromium setup:

[image]

### What did you expect to happen

a thumbnail to click on and view details in all its glory

### How to reproduce

it’s probably hidden due to layout and not 404-ing

### Additional information

rpi 5, hdmi 1024x600 touchscreen rpi os

thijstriemstra, 2025-04-10:
it’s in the history:

[image]

pedrolamas, 2025-04-11:
Hi @thijstriemstra thank you for reporting this issue.

I believe this is a due to the layout media styling we apply, I will take a look and see if we can improve this.

thijstriemstra, 2025-04-15:
same applies to spoolman card (even on win desktop firefox): 

[image]