fluidd-core/fluidd #1703: Show z diff from base in SCREWS_TILT_CALCULATE menu

MrPota2, 2025-08-24:
### Is your feature request related to a problem? Please describe

When running SCREWS_TILT_CALCULATE the menu displays the z height instead of the diff. This makes parsing the actual difference difficult. 

Current menu:

[image]

### Describe the solution you'd like

I propose adding a diff from the base screw in parenthesis to aid in understanding the results.

Proposed solution:

[image]

Potentially with a total min max diff:

[image]

### Additional information

I am willing to do this myself if it is something other users would like have. Please let me know if you see any room for improvement on the design or idea.

pedrolamas, 2025-08-25:
Hi @MrPota2, thank you for sending this Feature Request.

I now have a draft PR implementing this, but I have to admit I am not fully sold on the value of this feature, or in the visual end result (as shown on the screenshot on my PR)

MrPota2, 2025-08-25:
That was quick @pedrolamas!

I see what you mean with the visual end result. Perhaps it belongs with the turn value in some way instead. As it is just a more detailed representation of that value.

As for the value of the feature. It would help me get a better sense of the state of the bed and evaluate when it is good enough.  When you are down to 1-2 minute movements you get at bit blinded by the clock measurements.

I am willing to think a bit more thoroughly on the visual aspect if you have reconsidered the value.

pedrolamas, 2025-09-03:
I made a small change to the display order, moving the diff value to the left so the text looks more aligned now!

[image]

PR #1704 (2025-09-04): feat: show relative Z in Screws Tilt Adjust dialog
Displays the relative Z position of each screw in the Screws Tilt Adjust dialog, making it easier to compare screw heights.

The relative Z is calculated based on the base screw's Z value and shown alongside the absolute Z in the UI.

## Screws Tilt Adjust dialog

[image]

Resolves #1703
closed 2025-09-04 by pedrolamas