fluidd-core/fluidd #1719: Show filament costs of print in history taken from spoolman

micharaze, 2025-09-24:
### Is your feature request related to a problem? Please describe

I have to calculate the total filament costs of a print manually currently, but it could be actually done by fluidd and spoolman. 

### Describe the solution you'd like

It would be great if the print history log could also display the total filament cost of each print.
The value could be calculated based on the amount of filament used and the spool price information provided in Spoolman.

### Additional information

Thanks a lot for your work, much appreciated!

micharaze, 2025-09-24:
@pedrolamas thanks for the quick reply. I created a moonraker issue on their github, lets see how this goes.

https://github.com/Arksine/moonraker/issues/1023

thijstriemstra, 2026-10-30:
> Print history is maintained by Moonraker, so this would first need to be implemented there.

if you take OrcaSlicer for example, it doesnt care about print history, it just takes the nr (cost) defined in the filament profile. This is also how fluidd should treat the cost, except take the cost from the spoolman api, not a filament profile. or am i missing something here?

pedrolamas, 2026-11-08:
Moonraker currently records the Spoolman spool IDs in the print history, but it doesn't indicate how much filament was used from each spool. This makes calculating costs for multi-spool prints nearly impossible.

So, while the single spool calculation can be done, I would argue that Moonraker should store the used spool IDs *AND* the length so we could provide a proper calculation.

Additionally, the spool cost itself is not stored in the print history, so any changes to the spool cost in Spoolman would alter the displayed total - though this is a minor issue.